ytai/ioio #118: Missing docs on PWM and peripheral pins

gwideman, 2016-01-21:
1. On page https://github.com/ytai/ioio/wiki/Getting-To-Know-The-IOIO-OTG-Board (and corresponding V1 page) there are comments about how PWM is available on some (all?) pins that are marked "P" on the back of the board. But there's no picture of the back of the board, and I found no other indication which pins these are. 

2. It would be helpful if this "P"-ness was documented in the pin function table on this page.

3. Is it indeed the case that any P pin can furnish PWM?  

4. The pin function table is very frustratingly formatted, being too wide to fit in the narrow page width, and requiring a scroll bar, which is waaaay at the bottom.  Maybe change CSS for .markdown-body table to overflow: visible, or use a smaller font size. Or if that's not possible, post a spreadsheet. Something readable and printable. 

Thanks.

ytai, 2016-01-21:
1. Here's a picture of the bottom: https://cdn.sparkfun.com//assets/parts/9/3/1/5/12633-03.jpg
2. The "P"-ness is what is referred to as "PPSO" in the table.
3. Yes.
4. GitHub have a nasty habit of changing the page layout and not leaving too much room for customization. I've raised the issue with them, but they don't seem to fix it. If you find a workaround, let me know (you can edit the wiki yourself, you know...)

closed 2016-01-21 by ytai

gwideman, 2016-01-21:
Thanks for quick response. Now I come to think about it, I posted an issue years ago about the oversized fonts on GitHub wiki, and clumsy table formatting (not to mention fixed page width). Very inappropriate for tech docs.  
Anyhow, I've reduced the font size on the table using the "sub" tag hack, to pretty good effect. Also narrowed a couple of columns that were wider than needed solely due to their headings.