ytai/ioio #119: IOIO OTG not recognized by Android 5.x devices

griggheo, 2016-01-28:
I am trying to connect via USB to a IOIO OTG board from Android devices running Android 5. I haven't yet been succesful. Here is what I have:

IOIO board:

Hardware version: SPRK0020
Bootloader version: IOIO0401
Application version: IOIO0506

Moto X phone running Android 5.1
Galaxy Tab A tablet running Android 5.0.2

I tried different USB cables, and I also tried putting the IOIO board in host mode. I also tried enabling/disabling USB debugging mode on the phone. All to no avail so far.

Any suggestions would be greatly appreciated!

ytai, 2016-02-01:
I'm using the IOIO regularly with a Moto X. I don't think what you're seeing is a systematic issue.
Some basic questions:
- Have you checked that the charge limiting pot is fully clockwise?
- In what way does it fail?
- Does the board work with any other devices? With a PC?
- Did you get the board from an official vendor?

griggheo, 2016-02-02:
Thanks for replying, Ytai. I have been trying to recognize the board (official, from Sparkfun) using the IOIO Hardware Explorer app and the Hello IOIO app. Both apps can see the board just fine in Android 4.x but not in Android 5.x. I don't have the charge limiting pot fully turned clockwise so I will try that.

ytai, 2016-02-02:
Also: did you turn USB debugging **off**? Does the OpenAccesory dialog pop whenever you're connecting the IOIO? Are you getting the charging indicator to show?

hbmartin, 2016-02-02:
I'm having the same problem on Nexus 7, regardless of whether USB debugging is turned on or off. I've tried requiring it to run only in OTG, and only in AOA. In neither case does the looper's connection handler ever get called.

hbmartin, 2016-02-02:
* The hardware I have is "SPRK0020"
* I've made sure USB debugging is turned off, and there is still no dialog popop showing.
* I have tried it in both A and H mode with the same results.
* The Android device is being charged. I'm running a 10V / 4A power supply
* Charge pot turned all the way clockwise (max)
* I'm not even seeing any system notification (in wifi logcat) that there has been a USB device connected, only a power device.

mohsenpashna, 2016-02-02:
Have you tried and is it working with BT dongle?

ytai, 2016-02-23:
Nexus 7 works for me. You should disable USB debugging. Enabling ADB over
WiFi is NOT sufficient.

lattyak, 2016-02-29:
I having the same problem on an S4 running 5.0.1. Same app works fine an S2 (4.1.2).  Tried all combinations of H/A switch, USB debugging on/off, Developer options on/off. BT works on both. I recompiled the IOIO Libs to add debugging.

UsbManager.getAccessoryList() always returns null, as if the IOIO is either not recognized as an Accessory device, or not matching the filter. Is there a change in 5.x related to AOA filters? The filter in the IOIOLib is &lt;usb-accessory model="IOIO"/&gt;.

For Device connection, the app is listening on port 4545, but no connection is made from the IOIO. Did 5.x add a firewall?

sibomots, 2016-11-04:
Similar problem on IOIO-OTG

Specs:

```
C:\code\IOIODude-0102>ioiodude --port=COM3 versions
IOIO Application detected.

Hardware version: SPRK0020
Bootloader version: IOIO0402
Application version: IOIO0506
```

Host platform:
WIndows 10
Android Studio ( Version 2.2.2)
Android handset Moto G (OS = Android 6.0.1)
JDK 7 and JDK 8 (both tried)

Process:

Built HelloOIOI and IOIOSimpleApp directly from sources on github.
Success after gradle/InteliJ reporting needing to upgrade stale versions of the SDK.  All sucessful, builds cleanly.

Am able to provision the hanset with the app.  App runs on the handset.

Issue/bug:

In both cases, the setup() function in the app never fires, the UI is never enabled.

Android handset is running 6.0 Android.   Connected between Android handset and IOIO board via USB.

What should I be looking for to troubleshoot the problem of the UI not being enabled in the app (due the connection between the App and IOIO not established)?

Also tried with apps built via JDK 7 and JDK 8 (some older post/wiki mentioned the suggestion to use JDK 7, but this didn't resolve the issue).

Ideas?

-thanks

ytai, 2016-11-04:
First things to look at are whether the phone charges and whether the accessory dialog pops up when connecting the IOIO. This is completely independent of anything installed the phone.

sibomots, 2016-11-04:
OK.  Well that was embarrassing.

I should have checked the power supply.  I had a weak battery.  Enough to illuminate the power LED, but not enough to drive the system.  I'm not entirely sure.  This time I'm using a bench power supply instead of a battery.

All is OK.  Good.  Nice system. Baring my stupid mistakes we may just get this thing off the ground.

AGKhalil, 2017-02-14:
Hey all,

I'm running 6.0.1 Android and the phone's not recognizing the IOIO OTG board at all. No accessory dialog shows up and the phone doesn't charge. Do you have any suggestions as to what could be wrong? I always made sure the USB debugging is off and I'm using my Arduino as the power source, 5V.

Any help would be very appreciated.

Thanks,
Ahmed

ytai, 2017-02-14:
Use a proper power supply?

AGKhalil, 2017-02-14:
Hey Ytai,

Thanks for responding. I tried a 9.6V RC battery and still nothing. If it helps, I am using Huawei Honor 5X for my phone and I upgraded the IOIO firmware to IOIO0506.

Any thoughts?

Best,
Ahmed

ytai, 2017-02-14:
I wondering if this is specific to this phone. Have you tried this exact
setup with a different phone?

AGKhalil, 2017-02-14:
Yes and none of the work either. I've tried a different android versions too 4, 5, and 6. Nothing works.

ytai, 2017-02-14:
OK, so we have no reason to believe this has anything to do with a
particular Android version, but rather something about your specific setup.
Can you provide some details about how you're connecting everything, is
this a IOIO from an official vendor, etc?

AGKhalil, 2017-02-14:
This is a IOIO I purchased from Amazon, the link is below:
https://www.amazon.com/Sparkfun-PID-13964-IOIO-OTG-Headers/dp/B01M7O5Z9N/ref=sr_1_4?s=electronics&ie=UTF8&qid=1485529441&sr=1-4&keywords=ioio+board 

The image below should show you how everything is connected.
[image]

Im us running MacOS Sierra 10.12.2. Let me know what you think.

ytai, 2017-02-14:
> Headers/dp/B01M7O5Z9N/ref=sr_1_4?s=electronics&ie=UTF8&qid=
> 1485529441&sr=1-4&keywords=ioio+board

Is the charge current limiter (the little pot on the IOIO) fully clockwise?

lattyak, 2017-02-14:
The IOIO sends an extra USB message indicating it supports Accessory Mode. However, this is an optional message, and if it gets corrupted the Android does not ask for it to be resent. And in 5.0 a change was made to ignore all Accessory Mode messages after a timeout. I found older versions of Android it would eventually connect via Accessory, while for >=5.0 if the first Accessory msg was dropped it would never connect. The IOIO code should probably send out extra Accessory Mode messages blindly.

Why did the first Accessory Mode message get corrupted. I found 2 culprits causing this. First was a bad connector. During the insertion, noise can be generated on the data lines because of poor plating on the contact surfaces. Dirt, contact pressure, tolerances and wear can also affect this. I went through lots of different USB cables, and found some worked much better than others. Also phones with a worn USB connector were more problematic. Why does it only affect the Accessory Mode message, well it is the only message the Android side does not ask for a retransmission if there is an error.

The second issue was power supply to the IOIO. The USB charging protocol increases the charging current till the USB voltage drops below something like 4.75V, then it backs off the current till the voltage rises. It's a little more complicated that than, but it's basically testing for a current limit. The IOIO is designed to work on 5V (its DC-DC convertor has a low drop bypass MOSFET when input is 5V), but that makes the system run on ~4.85V -- not much headroom for the charging protocol. Also when it hits the current limit the rest of the system is affected.

I ended up using a 12V supply because it is the most efficient voltage for the IOIO's DC-DC convertor. Also you can buy cheap 12V surveillance camera power supplies that have the same size barrel connector. Anything above 8V will probably work as long as it can supply enough surge current for the charging protocol.

-John

AGKhalil, 2017-02-14:
It works now! It was the charge current limiter. Thanks a lot Ytai and John. The HelloIOIO app works flawlessly now too.

ocenandor, 2021-02-24:
Hello everyone!
I try to connect my IOIO board to my phone Vivo y81 and it doesn’t work.
IOIO works normally when it’s connected to PC and managed via PC command line. It also works via bridge. (I launch HelloIOIO app)
When I connect my IOIO board to the phone it’s charging but no messages about IOIO appear. I charge it with 9V battery in Vin and Gnd pins. The charge current limiter is fully clockwise.
Please, help me

topherbuckley, 2021-03-17:
@ocenandor are you using an Android 5.x device? If not, please open a separate issue and include all the details of your phone including OS version, and the steps you take to prior to connecting the IOIO board to the phone. Have you followed the Sparkfun tutorial? If so, mention which steps you have completed and which you are stuck on and hopefully I can recognize some common pitfalls. Regardless, this discussion should be on another issue if it is not particular to Android 5.x devices. Cheers.