ytai/ioio #120: did i missed the how to install libraries ?

FrankOcean11, 2016-02-24:
i downloaded the android studio and i do not see anything thing that says motor control or video capture or openCV  for visual servoing "this is the main reason i downloaded the Android sdk"  so that i can use the ioio with my pc "pc mode" to run serial commands to servo according to video feedback "autonomous"  i think i when through every note in this gihub in search of such approach and all i see is people on youtube using this methods

ytai, 2016-02-24:
There is nothing about the IOIO that is specific for computer vision or visual servoing. The IOIO is merely a tool that allows you to interface electronics such as motors and sensors to a program running on your PC or Android device. A lot of people have indeed used it in robotics application, but the IOIO is merely one piece of such applications, independent of other systems such as perception and motion control.

closed 2016-02-24 by ytai

FrankOcean11, 2016-02-24:
ok so i own arduino , do i even want to own this board as well ? i dont see much of a community backing this up specially in robotics, what i do see is a bunch of show offs who aint about to tell  their setup secrets ,

ytai, 2016-02-25:
I'll try to ignore the offensive tone and answer to the best of my ability (but mind you, this kind of language is usually not the best way to get people to go out of their way to help you out).
The way people usually use the IOIO is by connecting it to an Android device (phones/tablets) and leveraging all the advantages of having a full blown Android device in the application, namely the fact that you have a powerful mobile computer with elaborate connectivity, sensors (camera, GPS, IMU, ...), display, etc., all with easy to use APIs and good development tools. The IOIO extends this system by adding the ability to easily connect devices and circuits that are not part of this package called and Android device, and makes it similarly easy to communicate to those external peripherals.
Many software libraries for Android are readily available, including OpenCV.
Likewise, if you need this ability to connect low-level (as opposed to, say, USB) peripherals to your PC and communicate with them using a Java API directly from the PC, the IOIO would do that for you. In many cases, you can solve the same problem with an Arduino. One big difference would be that in the latter case you'd need to write two pieces of software (PC, Arduino) and design the communication interface between them yourself, which may be easy or hard depending on your skill and you application's complexity. You'd also have lower bandwidth on most Arduinos in comparison to the IOIO. And again here, open-source libraries for robotics are available for a Java environment and are completely independent of the IOIO and thus outside the scope of the IOIO project.
As far as people sharing how they're implementing their IOIO-based projects, some people share, some people don't. I usually share my "setup secrets" as you can see on my Github account and my blog, but there's certainly no must to do that, just like is the case with Arduino. Arduino happens to be a more popular platform, so you tend to find more open source code written for it. Nevertheless, for some applications, the IOIO would have a lot of advantages.
And really, use what you please. The IOIO is not a big business, it is my personal contribution to the Maker community of which I'm part. I'm happy it is helpful for some, I'm not offended by the fact that it is not helpful for others. The most important thing is that you make something cool, enjoy making it and learn something in the process. That's all there's too it.
Take care!

FrankOcean11, 2016-02-25:
ok first of im glad you chose to ignore what was not intended to offend you, i speak of those who dont share and obviously your not part of that crowd.  Now your telling me that this device is your feat and you dont have any open source applications where your ioio is being used in conjunction with a camera to drive motors? 

i liked to use ioio to control the jaw diy servos and the video functionalities of the phone for vision 'machine vision.
[image]

again my apologies i did not meant to offend who im asking for help, its just frustrating when your trying to make something not being done by the community 'my AI Android' and while it sure is on that list you would call 'cool' i dont see anyone even sharing their methods.  Like i said i been using arduino and i have no reason to use the ioio other then to learn but even with the arduino i started with small example code and a easy to use and install .exe ide program which i dont find that to be the case with ioio.

ytai, 2016-02-26:
In order to understand you better, can you please explain to me what your Arduino-based development environment looks like? Obviously, you're not running any Vision code on the Arduino itself, so what does your *other* program look like? What environment are you using for it? How do you make your Arduino code talk to your vision code?

FrankOcean11, 2016-02-26:
i use 2 protocols wireless for the limbs "they are modular" the nervous system i communicate via optic connection and mcu use rs232 /UART.  Some go through the USB others through a Server, it really depends on the speed and needed and what module might be relaying information 'tx/rx' and i use Visual Studio2013