ytai/ioio #125: Can't connect to IOIO-OTG with Studio compiled app

barehill, 2016-07-20:
Am trying to implement a copy of the simple HelloIOIO and HelloIOIOService apps compiled using Studio. The apk's supplied in the V5 download both work. However compiling each within Studio fails on both. The connection Toast shows recognition when the cable is plugged in. But the app's don't operate the yellow LED nor show Service connect. I carefully checked the manifest files and copied code from the examples to my Studio projects. Any ideas? I have many hours on this now.

Droid Moto G 4.4.4
IOIO-OTG
Firmware IOIO0506
Bootloader IOIO0400
Hardware SPRK0020

ytai, 2016-07-20:
How did you create your source tree? Can you provide the build output?

barehill, 2016-07-21:
Ytai,
Thanks for the quick reply.
Are these the trees you want?
IOIO:
[image]

IOIOService
[image]

ytai, 2016-07-21:
Doesn't seem like you're trying to compile the provided example apps (at least in the later case). Can you please try to extract the source code from the software release bundle and build it unmodified?

barehill, 2016-07-21:
Thanks, I'll try that. 
But I thought I could copy source files into a new project and get it to work, as a first start to developing my own new code.

ytai, 2016-07-21:
Yes. of course you can, but something is apparently missing. So let's take a step-by-step approach by first verifying that you can build and run the unmodified example sources.

barehill, 2016-07-21:
Ytai,
Yes, both HelloIOIO and HelloIOIOService work by importing them from the V5 examples and then compiling and loading the apks. Great. But how to get the equivalent code into my own project? That's where I seem to have the problems.

barehill, 2016-07-21:
By the way, I worked with the initial IOIO boards some 4 years ago while trying to develop an autopilot for my boat, a 44' trawler style cruiser.  But I had to give it up due to issues with the boat 12V system - spikes destroying the IOIO regulator, and issues with the flyback voltage from the servo clutch on the boat drive system. But those are resolved and I don't want to pay $1300 for a commercial autopilot head. So I'm back at it. :)

barehill, 2016-07-23:
OK, got the HelloIOIO example code copied to my Project, now working and off and running. Still don't know why the HelloIOIOService couldn't be copied but I'm not an issue right now.

closed 2016-07-23 by barehill