ytai/ioio #126: Partial success

MrAda, 2016-07-28:
Hello,

I am not a Java programmer (C/C++/Ada is what I am good at) and I have manage to cobble an app to turn on my relays for my project.  I have several questions:

I'd like to work with the I/O pins, both reading and writing, and it seem that I can either read or write with them.  Has anyone done read and write on any of those pins?  I am using 11 - 14.

I tried to build the torture test and I got:
Error:(48, 0) No such property: OSSRH_USER for class: org.gradle.api.publication.maven.internal.deployer.DefaultGroovyMavenDeployer

I noticed that I see references to a yellow LED on the board in the examples.  Mine is red so is there a yellow LED somewhere or did my manufactoer just give me a red LED?

I tried a distance BT test, trying to get as best a LOS and I noticed that I can only broadcast.  My range was about 35m.  Is that typical?

Thanks,
Chris

ytai, 2016-07-29:
1. Any of the pins can be used as digital input or output. Please refer to the IOIO wiki that has all this information.
2. You're probably attempting to build the app from a source tree cloned from github. This is not the intended use. Please download the software bundle from the downloads page on the wiki, which has the source tree that is intended for users to build.
3. The red LED indicated power. The yellow LED is under software control.
4. Don't quite understand your comments regarding Bluetooth, please be more specific.

closed 2016-07-29 by ytai

MrAda, 2016-07-29:
1> I was under the impression that certain pins could be use for both input and output. I guess not.
2> Yes I did grab that from the cloned github.  I have download the java app so that is what I am using to try different things out.
3> I guess all of the examples that claim to toggle the yellow LED is really toggling the red LED instead.  I thought maybe my board was slightly different for some reason.
4> The time it takes for the IOIO's to connect to my phone using BT is taking maybe 15 seconds or more.  I don't know if that is normal behavior or maybe it is my dongle.  I am going to have replace it because of the distance as I need more than 30 m.

ytai, 2016-07-29:
1. Not sure what you mean by using the pins for both. If you mean using them in open drain on a shared bus while reading from the bus, this can easily be accomplished by tying two pins together and using one as input one as output.
2. Does that build OK for you?
3. Sounds like you have an unofficial clone. Be warned that some of those were found to be of very poor quality.
4. 15 seconds is not normal. Might be an issue with the phone or dongle. Another possible issue is that you're trying to connect to multiple IOIOs while some of them may not be available. Some phones' BT stack performs very poorly while trying to establish a connection to other devices.

MrAda, 2016-07-29:
1) I'd like to know the state of the pin as well as being able to set it.  I am still trying to wrap my  head around this "open drain" idea.  But what you suggest sounds interesting, just have to understand it better.
2) Yes.  I have been doing ok with the JAVA examples.  I am just trying to build upon those examples to see if I can make an app that I can use for my project.
3) I hope not, but that is something that I thought.  Here is the auction on ebay:http:
//www.ebay.com/itm/282051402395
Maybe you can tell me if I did ok with this purchase or not.
4) Well in my situation I am only connecting one at a time right now.  But you are right in that I will be using two eventually.  I am thinking the dongle is cheap so that could be the issue.

ytai, 2016-07-29:
1. If you're working in normal (push-pull) mode, there's no need to ask the pin what its state is. Whatever you set it to, that's what it is. You can easily store this state in your app.
3. Knock off... If you want an official one that's verified to be tested and manufactured well, buy from SparkFun or SeeedStudio. You'd also be getting an improved version that a more reliable power supply circuit.
4. If more than one is *paired* (i.e. you see more than one IOIO* device in your BT settings page) the library will constantly try to establish a connection regardless of whether or not that IOIO is actually present, so make sure only one is paired in order to test whether this is indeed the problem.

MrAda, 2016-07-29:
1) ok.  I am saving the state anyway for the open drain way to go anyway.  I'll look into the other way.
2) So I guess this is what I get for trying to save $10.00/unit I bought.  I feel a bit upset at the moment.
3) well given that I think I got bogus pieces of hardware (dongles included) I am unsure what I am going to do next.  I am trying to cobble a solution to my high demand usage on my A/Cs and now is the time to get it to work given the weather.  maybe I can use my RPi and one of the IOIO's instead of two being controlled by the RPi.  Seems to me that the Java doesn't won't work too well with two apps running (one to talk to each IOIO device) as it disconnects immediately when you go to another app.

ytai, 2016-07-30:
You don't need two apps to talk to two devices. You can have one app. A separate IOIOLooper will get created for each, each on its own thread.
If you're going to use an RPi, the Bluetooth behavior is likely to be different than in the Android case, possibly better.

Waged, 2016-10-06:
Hello Dr. Ytai  
I'm trying to build the HelloIOIO app and i got an error : 
Error:(48, 0) Could not get unknown property 'OSSRH_USER' for object of type org.gradle.api.publication.maven.internal.deployer.DefaultGroovyMavenDeployer.
<a href="openFile:C:\Users\l\Desktop\ioio-master\ioio-master\software\build.gradle">Open File</a>

and i have no idea how to fix it ... 
please help ... 
thanks Dr.

ytai, 2016-10-07:
Please build from the software bundle that you download from the downloads
page and not from a git clone.

Waged, 2016-10-07:
Thanks for help .. 
now i have error on my android phone when start to connect with IOIO ...
Incompatible firmware version IOIOLibIOIO0507
application firmware : IOIO0330
Bootloader firmware:IOIO0400

ytai, 2016-10-07:
Upgrade the firmware in the IOIO. There are instructions on the wiki.