ytai/ioio #127: New sensors to try with IOIO

MrAda, 2016-07-29:
Hello,

I bought these two sensors:
DHT22/AM2302 Digital Temperature Humidity Sensor Board
US-015 Long Range 700cm Digital Ultrasonic Arduino Distance Sensor Module

Has anyone interfaced these with the IOIO board?

guillesupremacy, 2016-07-29:
For temperature sensing, I used the BMP180 sensor. I made a translation of
its driver from C++ to Java.

https://github.com/guillesupremacy/BMP180_IOIO_Driver

Atentamente,

Guillermo Guzmán Sánchez
Móvil: 301 359 09 65

MrAda, 2016-07-29:
I am thinking of translating code from Java to C++ using QT as I don't care for the android studio.

ytai, 2016-07-30:
Please use the ioio-users Google group for such questions. The issue tracker is intended for bug reports or feature requests. Thanks.

closed 2016-07-30 by ytai