ytai/ioio #130: IOIO Android 6 Issue

ShaunFinch, 2017-01-23:
Hi I am unable to get Adroid 6 to link with my IOIO.

I have tried two Android devices with several IOIO boards. I have turned off debugging. The IOIO have latest App-IOIO0506 firmware installed. When connected the Android device displays the charging symbol.

All the IOIO I have tried link up to and older device I have running Andriod 4.  
I've read though the posts on here with a similar question about Andriod 5, tried all suggestions with no luck.

Can anyone help please?

AGKhalil, 2017-02-13:
Hello,

I am having a similar problem, but my IOIO runs the V3.30 firmware. When I connect it to my phone, running Android 6.0.1, nothing happens. I have made sure that debugging is always off when I connect them. Any help would be much appreciated.

Best regards,
Ahmed

ShaunFinch, 2017-02-15:
Ahmed - thank you for copying me in this message. I have not been able to
resolve the issue I have with IOIO and Android 6. Tried several IOIO and
Android devices. SO any help would be great.
 
Yours
Shaun
www.Kart-Crazy.co.uk <http://www.kart-crazy.co.uk/> 
02380 972010
 

  _____  

<https://github.com/ytai/ioio/issues/130#issuecomment-279541894> GitHub, or
mute
DCRk5Viks5rcNcPgaJpZM4LrJE5> the thread.
<https://github.com/notifications/beacon/AYH0uDlXMcTSN3NDsrlfDIgdgs5M_C6Cks5
rcNcPgaJpZM4LrJE5.gif> 

No virus found in this message.
Checked by AVG - www.avg.com <http://www.avg.com/email-signature> 
Version: 2016.0.7998 / Virus Database: 4756/13946 - Release Date: 02/13/17

ShaunFinch, 2017-02-21:
Hi did you get a reply to this?
 
Yours
Shaun
www.Kart-Crazy.co.uk <http://www.kart-crazy.co.uk/> 
02380 972010
 

  _____  

<https://github.com/notifications/beacon/AYH0uDlXMcTSN3NDsrlfDIgdgs5M_C6Cks5

AGKhalil, 2017-02-25:
Hey Shaun, yes I did on a different thread. I attached the link below:

https://github.com/ytai/ioio/issues/119#issuecomment-279816569 

Best regards,
Ahmed

brtietz, 2017-04-25:
I wanted to weigh in since I am experiencing similar issues. Here is what I have observed:

Android 4.4 over adb: works
Android 4.4 over OpenAccessory: works
Android 6.0 over adb: works
Android 6.0 over OpenAccessory: does not connect

I have not seen a permissions dialogue come up on any Android 6 device that I've tried (
Lenovo TB3-730F ZA11, specifically). Has anyone else had permissions issues on Android 6 and above?

brtietz, 2017-06-30:
For anyone else debugging this, we've found a useful tool for determining if your tablet supports open accessory:

https://github.com/gibsson/linux-adk

We've established that our Android 6 tablets can connect over open accessory at 5 V using the following procedure:

(0) disconnect everything
(1) reboot the tablet
(2) plug the IOIO board into the power supply
(3) plug the tablet into the IOIO board
(4) repeat 2 & 3 several times

It seems like the suggestion in #119 about "The IOIO code should probably send out extra Accessory Mode messages blindly." would be helpful.

andreassh, 2017-08-21:
Regarding @brtietz answer:
"I have not seen a permissions dialogue come up on any Android 6 device that I've tried (
Lenovo TB3-730F ZA11, specifically). Has anyone else had permissions issues on Android 6 and above?"
I've tried connecting through with the same Lenovo TB3 tablet, and I get no permissions dialog neither. Can't seem to establish a connection through neither my app (which works fine on a samsung galaxy with android 5) nor with the HelloIOIO app. Did you ever get it to work over OpenAccessory?

brtietz, 2017-08-24:
@andreassh Yes, including on that specific tablet. The trick was to reset the channel if we didn't connect within a certain number of cycles (akin to blindly sending out more Accessory Mode messages). We have the code on a fork here: https://github.com/catalia-health/ioio

We haven't put in a pull request just yet because we're still dealing with a few edge cases. If our application exits and resumes (as would happen for a software update), we lose connectivity until reboot. I'll try to remember to post here when we're ready to upstream the code.

100TWH, 2018-04-23:
@brtietz I'm facing the same problem with no AOA popup and would like to try out your fix but I have never done any IOIO firmware programming so I appreciate any help I can get. From what I understand these are the steps I need to take:

1. Download the updated code, I think only firmware/app_layer_v1/main.c is affected, right?

2. Build a new IOIO application firmware using MPLAB-X following the instructions here:
https://github.com/ytai/ioio/wiki/IOIO-Developer-Getting-Started-Guide

3. Re-program the IOIO OTG board using IOIODude following the instructions here:
https://github.com/ytai/ioio/wiki/IOIO-OTG-Bootloader-and-IOIODude

Is this correct way to go?

If you or someone else already has an already complied application image (*.ioio or *.ioioapp) with the AOA fix, I will be more than happy to use that :)

Best regards,
JP

brtietz, 2018-04-27:
Yes, those are the correct steps.

100TWH, 2018-04-27:
I followed the step and managed to build the customized firmware and flash it to my IOIO but when I attached it to my phone (a LG Nexus 4 running Android 5.1.1) the recharge icon started to rapidly go on/off and just continued to do so. The IOIO was not identified when running for example HelloIOIO. The trimmer was at its max. Didn't have time unfortunately to debug it. I flashed the stock app firmware again and then everything was back to normal.

@brtietz Is this something you have seen before? I think that you made a note somewhere that the code was not ready to be pushed due to some remaining issues?

brtietz, 2018-05-21:
Sorry for the slow reply - that's not something we've seen here. We've (mostly) worked around the issues I mentioned on the Android side, but haven't made an example in our fork yet. That said, having the code do a hard reset on itself every 1.5 seconds if its not connected is a pretty sloppy solution. It'd be nice to upstream something a little more deft.