ytai/ioio #133: How to increase the charge current (more than 500 ma)

adi1133, 2017-05-17:
I am using a IOIO-OTG - V2.2 in host mode (OTG) connected to an Android tablet.
The ioio is powered from a 5V 2 amp wall charger, and the tablet is powered from the ioio.
I have checked the [schematic](https://cdn.sparkfun.com/datasheets/Dev/Android/IOIO-OTG_v22b.pdf) for the ioio and it looks like it contains no hardware for limiting current (only a 750 ma fuse).

I think the charging current is limited by the tablet to 500 ma because:
- The charge current trimmer is set to maximum
- When the trimmer is set to maximum, there is not effective current limiter
- The 750 fuse is intact (after days of usage)

I think the microcontroller inside the ioio instructs the tablet to not draw more than 500 ma.
The tablet requires more current to prevent it from discharging, how can I increase it ?

ytai, 2017-05-17:
I don't think that that's possible. The USB standard (and not the IOIO)
requires the device to draw >=500mA from the host. "Fast" chargers usually
work around this issue by using some non-standard tricks to let the device
know that they are a fast charger, but those tricks are generally mutually
exclusive with being able to exchange data with the host. They are also
device-specific in their nature.

Your options that I can think of are:

   - Get a different tablet that has a separate charging port or that
   requires less current.
   - Change your app such that it results in less current draw, such as
   turn the screen off etc.

adi1133, 2017-05-17:
Thank you for the fast response ! Have a nice day :)

closed 2017-05-17 by adi1133

iwl, 2019-04-27:
It is possible to draw up to 1,5A from an USB-Port, specified as Charging Downstream Port (CDP) in USB-Battery-Charging Spec, there is the usual data connection on that port also, in contrast to the Dedicated Charging Port (DCP) providing only 1,5A power without a data connection.
As fare I understood up to now there is a special handshake on D+/D- lines before the data connection is set up, may be this can be implemented in IOIO with a firmware change.
If there is a 750mA Fuse it needs to be bypassed of course.
I don't know if spec requires to make the bypass switchable for devices not using Charging Downstream Port protocol or no fuse is just fine.
There are probably chargers providing 1,5A or even more whatever you plug into.

lattyak, 2019-05-02:
Your best option is building a Y cable. Where only the data lines connect to the IOIO. And add some pull up resistors to trick the phone into drawing more current.

The IOIO has a LDO current limiting circuit. The only other option is modifying the components to allow more than 500 mA.