ytai/ioio #136: Possible to upgrad V1 via PC?

beniroquai, 2017-07-06:
I'm pretty new to this topic and accidentally bought the IOIO v1 (not OTG :/ )
Is there any possibility to update it via a PC? I have no Android cellphone <4.2 laying around. Are there any other chances? 

Many thanks in advance! :)

beniroquai, 2017-07-07:
Hey! Thank you very much for your reply! Short question: Could I install Android in a virtual machine on a X86 machine and try to access it from there? The other IOIO to programm has to has a higher firmware, right? I have two (same series, v1) which I cannot access from my device .. Firmware too low..

ytai, 2017-07-07:
> Could I install Android in a virtual machine on a X86 machine and try to access it from there?

You can use the Android emulator included with the developer tools and IOIOBridge (over Bluetooth only in the case of IOIO v1).

> The other IOIO to programm has to has a higher firmware, right? 

No. The feature that allows programming one IOIO with another got introduced very early on. I can't think of a reason why you wouldn't be able to use the IOIO v1 with your device or how a firmware upgrade would fix this, unless you have firmware that's 3-4 years on it, which is unlikely.

beniroquai, 2017-07-10:
Thank you very much! I think the FW version is too old. The phone gives me the message "The attached IOIO is running a application firmware version that is not compatible with the programmer. Expecting version IOIO0200 or higher" once I connect it. It's running Android 7.0 - but I think it's a problem with the FW of the IOIO itself, right? I bought the two IOIOs on Ebay. Probably they are fake ones? 

Is there any solution to flash new FW without a PicKit? Could I use the method with an Android Phone <4.1 (in case I could get one)? 
Is it worth searching for a developer kit for Android 4.1? 

All the Best!

beniroquai, 2017-07-11:
I was successfully able to connect the IOIO to the PC over Bluetooth using the Dude. It gives the following values:

                                                                                                                        `C:\Users\user\Downloads\IOIODude-0102>ioiodude --port=COM7 versions                                        IOIO Application detected.                                                                                                                                                                                                                      Hardware version: SPRK0016                                                                                              Bootloader version: IOIO0311                                                                                            Application version: IOIO0400     `

I'm wondering, why the Android App cannot and gives the versioning error. Any idea? Can I somehow fix the upgrading problem?

Best
Bene

ytai, 2017-07-11:
It's indeed fairly old. I think the problem is that the latest version of
IOIOManager is actually expecting a v5 IOIO.
Try an earlier version from here:
https://github.com/ytai/ioio/tree/master/release/apps

beniroquai, 2017-07-15:
Hmm. Thanks for the advice! I downloaded and tested all of them on an Nexus
5 and Sony Z5. Both of them didn't recognize the device (either Bluetooth
nor USB). Well. I think, that I cannot use it then. Maybe I have some luck
with the v2/OTG.

Best
Bene

ytai, 2017-07-16:
I'd recommend buying from a trusted source too (SparkFun or SeeedStudio)

beniroquai, 2017-07-22:
Did that! Thank you :)
Any way to use the old one somehow? I cannot connect to it at all..:/

ytai, 2017-07-22:
No reason why you shouldn't be able to.

beniroquai, 2017-08-07:
Hey. I've bought a PicKit 3 to flash the two IOIOs with the latest firmware. I'm following this video:
https://www.youtube.com/watch?v=s7YzkOFOvQk
Everything seems to work - correct wiring, etc. I'm able to read and erase. But writing the Firmware for the 206DA version (SPRK0015.hex) I'm getting an error:

Adress 0x0, read 0x40200, expected 0xFFFFFF 

Any ideas? Can I somehow make sure, that the PIC isn't burned already? 

Thank you very much again! :)
Best
Bene

ytai, 2017-08-08:
Can you attach the full output of the PICKit program as well as a photo of
your connections?
What this seems like is a failure to communicate between the programmer and
the IOIO.
Also, keep in mind that SPRK0015 is for a very old version of the IOIO, you
probably have the SPRK0016. The way to tell is by checking whether the PIC
is a *128DA* or a *256DA* model, respectively.

beniroquai, 2017-08-08:
Thanks a lot for your reply. Actually I have no idea, how I can get any information out of this PicKit3 Standalone App... But I've uploaded an image here:
https://www.dropbox.com/s/1bnqrmiez5mn4tw/2017-08-08%2013.42.18.jpg?dl=0

I've followed your explanation and the one on youtube. I guess all the wires are ok..The stat led is not blinking though. I have two IOIO with the  PIC24FJ256DA206 chip on it. Should be ok, right? I was trying to flash the 0016 file as well. 

Thanks a lot again! :) 

Best
Bene

ytai, 2017-08-08:
Those long wires and there fact that they are split in the middle might be
a problem. Try to avoid this if possible. Send a screenshot of the app.

beniroquai, 2017-08-09:
Awesome! Shortening the wires did the job! :)))
Now I wanted to connect to it using the USB-connection/Bluetooth connection. The playstore App always says "Waiting for target"..This means, it's not finding it, right? Android suggests to open your app, so the device is being detected I think. 
Which firmware version would you suggest to use with the latest Android app (latest Android OS also).

I hope I can get at least one of the IOIOs to work. The other one is fried I guess...;) 

Best
Bene

ytai, 2017-08-09:
Did you follow the pairing process, as per the wiki?
Also, which app are you referring to? I recommend testing with HelloIOIO,
provided in the software bundle.

beniroquai, 2017-08-10:
Nice! I can now control the device using Blutooth. There was an app to switch on/off the LED. Awesome! Lets start hacking! :) 
Thank you soo much for you help. Will order some new ones, but this time from the correct source with OTG support! 

All the best and thank you for putting so much effort into this project! :)

closed 2017-08-10 by beniroquai