ytai/ioio #137: Problems compiling 'Blink' project

GenericWorkName, 2018-01-11:
I've been trying to compile the 'Blinky' code example for a while now.
After setting up MPLAB with the XC16 compiler (which, according to microchip, is equivalent to the X30) I get the following compilation (or linkage) erorr:
`Options have been disabled due to restricted license
Visit http://www.microchip.com/MPLABXCcompilers to purchase a new key.
"/opt/microchip/xc16/v1.33/bin/xc16-gcc"   -o dist/IOIO0021/production/blink.production.elf  build/IOIO0021/production/main.o      -mcpu=24FJ128DA106        -omf=elf -DXPRJ_IOIO0021=IOIO0021  -no-legacy-libc    -Wl,--local-stack,,--defsym=__MPLAB_BUILD=1,,--script="../common/dummy.gld",--no-check-sections,--data-init,--pack-data,--handles,--isr,--no-gc-sections,--fill-upper=0,--stackguard=16,--no-force-link,--smart-io,--memorysummary,dist/IOIO0021/production/memoryfile.xml,-T../blapi/app_IOIO0021.gld 
/opt/microchip/xc16/v1.33/bin/bin/../bin/elf-ld: Warning: linker script did not specify CRT_STARTMODE file, default: crt_start_mode_normal
dummy.gld.01:1269: undefined symbol `__reset' referenced in expression
make[2]: *** [dist/IOIO0021/production/blink.production.hex] Error 255
make[1]: *** [.build-conf] Error 2
make: *** [.build-impl] Error 2
nbproject/Makefile-IOIO0021.mk:135: recipe for target 'dist/IOIO0021/production/blink.production.hex' failed
make[2]: Leaving directory '/home/user/ioio/clones/ioio/firmware/blink'
nbproject/Makefile-IOIO0021.mk:90: recipe for target '.build-conf' failed
make[1]: Leaving directory '/home/user/ioio/clones/ioio/firmware/blink'
nbproject/Makefile-impl.mk:39: recipe for target '.build-impl' failed

BUILD FAILED (exit value 2, total time: 962ms)
`

Cherish-Gww, 2022-02-16:
I have install XC16 compiler v2.0.0, v1.7.0, v1.6.0, v1.4.1, v1.3.3, v1.3.0, and v1.2.0.
Only v1.2.0 can compile blink project.

Cherish-Gww, 2022-02-16:
@ytai 
Are there any plans to update the firmware of ioio? For example, support python

hannesa2, 2022-02-16:
I guess not. @ytai lost the interest. We only can help ourself

Cherish-Gww, 2022-02-16:
@hannesa2 
OH GOD, PLEASE, NO.
Is there have a good open source project, that python can access mcu?
pyMCU project does not access.

hannesa2, 2022-02-16:
You mean this http://www.circuitsforfun.com/pymcu.html ? Sorry, I can't help you

topherbuckley, 2022-02-17:
@Cherish-Gww not sure what you are looking to do, but if you are just looking to use python with a microcontroller, you can try [micropython](https://micropython.org/). There are plenty of boards out there that can run this (e.g. Seeed Studio Xiao).

topherbuckley, 2022-02-18:
@Cherish-Gww is this a past use case of the IOIOBoard? I'm not familiar with such a use case and not sure if the IOIOBoard would be your best choice for such an application. Again, I'd recommend looking at other boards like the Seeed Studio Xiao as this has such an API built in already.

Cherish-Gww, 2022-02-18:
@topherbuckley 
Thank you very much, I found that mircopython has been provide pyboard script, so that running on PC python script can access mcu peripheral via pyboard object.
here is the doc: https://docs.micropython.org/en/latest/reference/pyboard.py.html