ytai/ioio #138: NUll pointer Exception in BluetoothIOIOConnectionBootstrap.java:68

pranaysharma, 2018-05-13:
Hi,
We there is a paired Bluetooth Device with no name then a Null pointer exception is obtained on line 68 as shown below

`for (final BluetoothDevice device : bondedDevices) {
				if (device.getName().startsWith("IOIO")) {`

`device.getName()` returns `null` in the above case thus throwing an exception.

y-a-n-n, 2020-08-19:
See fork here for a fix:
https://github.com/SmartMoveSystems/ioio

Created a PR but this project doesn't seem to be properly maintained

y-a-n-n, 2020-08-26:
Thanks. Is there a release incoming?

hannesa2, 2020-08-27:
We can do it. I'm thinking about to simplify build with jitpack.io but any suggestions are welcome

y-a-n-n, 2020-08-27:
I would second the idea of using jitpack as it's more streamlined. Happy to help with that endeavour. Should we raise an issue for that work?

topherbuckley, 2020-08-27:
@y-a-n-n maybe you can help me understand jitpack a bit better? I'm asking questions around this [here](https://github.com/ytai/ioio/commit/b27178dad8c513f84cce478c83542348155f94f7).