ytai/ioio #139: Unable to call StopService on IOIOService

brtietz, 2018-05-21:
I'm having an issue similar to #132 in which I will arbitrarily lose an IOIO connection after a few hours or a few days of being connected. While I'm still not sure of the root cause, I've been testing resetting my IOIOService subclass when it detected that a connection is lost. However, I get an App Not Responding error any time I try to call stopService:

"main" prio=5 tid=1 Waiting
  | group="main" sCount=1 dsCount=0 obj=0x7447b4e8 self=0xb4776a00
  | sysTid=20278 nice=0 cgrp=default sched=0/0 handle=0xb6f2cde4
  | state=S schedstat=( 6254942344 630620304 3353 ) utm=523 stm=102 core=1 HZ=100
  | stack=0xbe12f000-0xbe131000 stackSize=8MB
  | held mutexes=
  at java.lang.Object.wait!(Native method)
  - waiting on <0x0788d162> (a java.lang.Object)
  at java.lang.Thread.join(Thread.java:724)
  - locked <0x0788d162> (a java.lang.Object)
  at ioio.lib.util.IOIOConnectionManager.joinAllThreads(IOIOConnectionManager.java:90)
  at ioio.lib.util.IOIOConnectionManager.stop(IOIOConnectionManager.java:71)
  at ioio.lib.util.android.AndroidIOIOConnectionManager.stop(AndroidIOIOConnectionManager.java:95)
  at ioio.lib.util.android.IOIOAndroidApplicationHelper.stop(IOIOAndroidApplicationHelper.java:91)
  at ioio.lib.util.android.IOIOService.stop(IOIOService.java:134)
  at ioio.lib.util.android.IOIOService.onDestroy(IOIOService.java:92)
  at com.catalia.mabu.ioiomotors.MotorService.onDestroy(MotorService.java:159)
  at android.app.ActivityThread.handleStopService(ActivityThread.java:3244)
  at android.app.ActivityThread.-wrap21(ActivityThread.java:-1)
  at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1591)
  at android.os.Handler.dispatchMessage(Handler.java:111)
  at android.os.Looper.loop(Looper.java:207)
  at android.app.ActivityThread.main(ActivityThread.java:5728)
  at java.lang.reflect.Method.invoke!(Native method)
  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:789)
  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:679)

Does anyone have an example of a successful call to stopService, or is there a guaranteed mutex lock at this point in the code?

If it helps, I'm running on Android 6, and connecting via OpenAccessory.

WaterReNu, 2019-08-12:
Did you ever resolve this? we are having stability issues via usb, AOA and Bluetooth on Android 6. Previously in Android 4.4 no significant issues with USB.

brtietz, 2019-08-20:
Our stability improved by doing the following:
(1) Resetting our service every 10 seconds if a connection is not detected (using unbindService from the activity)
(2) Being extremely careful about threading while doing (1)
(3) Adding some null checks and exception handling in IncomingState.java and IOIOProtocol.java

I'll see if I can get (3) on the CataliaHealth fork (https://github.com/catalia-health/ioio) this week.

closed 2026-10-19 by hannesa2