ytai/ioio #143: Undesired preserving state of IOIO pins on android app closure

topherbuckley, 2019-05-08:
I'm seeing what I believe to be unexpected behavior. Two PWM pins I am using to drive motors reset to their expected 0 value after the USB cable is disconnected, but the values persist with a single non-zero value after the app closes (and USB cable still connected). 

I have the IOIO board hooked up to an Android 4.4 phone and it drives two wheels on a balancer robot via PWM. The PWM is either set to +/- 100% depending on the tilt of the phone and should reset to 0% when closing the app. 

The PWM, and wheels work as expected when the app is open, going back and forth without issue, though I see this strange behavior when closing the app. 

I've tried working around this via both the onStop method by setting the output to the PWM pins to 0, but this doesn't work as expected. It seems to only work the first time the onStop method run. 

I verified the onStop method is firing as expected with a Toast text, but setting the PWM pins to zero doesn't seem to work using the onStop method. It would seem the disconnect sequence from the IOIO library is not doing what I'd expect it to do (reset all pins to 0/float when app closes). 

I can replicate this by:

1. running the app
2. Seeing wheels move back and forth 
3. pressing the overview button
4. seeing the Toast text
5. wheels are stuck in one direction at set PWM
6. open the app again
7. nothing changes
8. press the overview button
9. seeing the Toast text
10. wheels continue spinning in stuck direction and PWM

I didn't see anything within the IOIOActivity.onStop that looks like it would cause this. Which methods are called when the USB is disconnected? Should I try to call this method within the onStop method or would this mess up some other shutdown processes?

techboycr, 2021-04-15:
As far as my understanding of the IOIO firmware goes, basically works as a FIRMATA. This means the app says "Pin 5 HIGH" and the IOIO board execute the action. 

Maybe this behaviour is due to not having a pull Down or pull Up circuit the IOIO board already implements pull down and pull up like this:

`private DigitalInput motorPin1_;
private DigitalInput motorPin2_;

motorPin1_ = ioio.openDigitalInput(6, DigitalInput.Spec.Mode.PULL_DOWN);
motorPin2_ = ioio.openDigitalInput(7, DigitalInput.Spec.Mode.PULL_UP);
`

Give it a try and lets us know if works.

topherbuckley, 2021-04-15:
Thanks @techboycr , but I have made other work around since posting this, and don't have time to dive back into it at the moment, but if there ever comes a time, I will be sure to reference what you have said.

closed 2021-04-15 by topherbuckley