ytai/ioio #144: Using Android Phone as Host

topherbuckley, 2019-06-24:
Using the IOIO Board as the host in an Android connection is quick and easy, as is using it in slave/device mode with a PC, but what is limiting it from being used as a slave/device when hooked up to a phone? 

I'd like to use the IOIO Board with an Android phone and make use of the existing phone battery rather than having to hook up an external source. If I use an OTG cable from the phone and feed the 3.8 - 5V voltage to a boost regulator to boost above 5V, can the existing android libraries handle this, or is there an easy way to adapt the PC libraries to work on Android? I can go digging into the code myself if no one knows, but I figure @ytai has already considered this and there is some reason why it is not openly supported.

ytai, 2019-06-24:
Yes, it is supported. See
https://github.com/ytai/ioio/tree/master/software/IOIOLibAndroidDevice

topherbuckley, 2019-06-25:
Thanks. Is this built into any of the demo applications? I'm not seeing this working in the default HelloIOIO App. The IOIO Board Powers on via my OTG Cable and phone, but no connection is ever established with any configuration of USB Debugging On/Off, connecting cables before/after app launches. 

Or is there documentation anywhere showing how to initialize the USB connection via IOIOLibAndroidDevice? I tried importing IOIOLibAndroidDevice to the default HelloIOIO project, and creating an instance of DeviceConnectionBootstrap thinking the onCreate method in there might be all I need, but this also did not seem to work. Should I be calling any particular method(s) from DeviceConnectionBootstrap or Streams for that matter in order to initialize this connection?

topherbuckley, 2019-06-25:
Turns out I simply had a faulty USB cable. Sorry for wasting your time :/ . I went back to trying with the IOIO Board as a host and this did not work, which led me to swap out the cables. It works great as both host and slave. We currently have around 20 IOIO Boards of various versions here at our lab and have been very pleased with them. Thank you for your efforts!

closed 2019-06-25 by topherbuckley