ytai/ioio #200: Release Notes

topherbuckley, 2020-09-16:
I don't see a way to comment on the releases themselves, but there isn't much info on the individual TAGs and releases. Is there a way to add release notes to these detailing what has been updated? It would likely take a bit more work to write these nicely for all beta releases, but at the very least, could we maybe just bundle all the master branch commit messages between the release and the previous release? For example I don't know if the newer releases are based on the AppCompat libraries or not without downloading the whole set and testing whether it breaks my project or not.

topherbuckley, 2020-09-16:
Found a better place to put this than issues. Moving [there](https://github.com/ytai/ioio/pull/193#issuecomment-693240751)

closed 2020-09-16 by topherbuckley

hannesa2, 2020-09-16:
You could improve [this](https://github.com/AppDevNext/ChangeLog) as a Gradle plugin , then it would be super easy

hannesa2, 2020-09-16:
The magic is here https://github.com/AppDevNext/ChangeLog/blob/master/generateTagGroupedGitlog.sh

hannesa2, 2020-09-16:
With the output of https://github.com/AppDevNext/ChangeLog/blob/master/generateTagGroupedGitlog.sh I can create the release here https://github.com/ytai/ioio/blob/master/.github/workflows/Android-CI-release.yml#L32

But I want to avoid boiler plate code, so a Gradle plugin would be the nicest way

topherbuckley, 2020-09-16:
Wow! You are WAAY ahead of me on this one. I'll take a look at that side project in the next fews days if I can get the time.