ytai/ioio #241: Android Compiling but no interaction with the board

techboycr, 2021-03-15:
Thanks for keeping this project alive!

I am trying without success to connect to IOIO-OTG board, the app compiles and installs correctly but there is no connection to the board. Is the same either in Host or A mode. The borad does charge the phone.

My setup is as follows:

- IOIO Otg with App-IOIO0506.ioioapp : board is working with the pc examples
- Samsung A10 running Android 10 
- Android Studio 4.1.1

This Samsung A10 seems to have support for Open Accessory:

`abrahampinedazelaya$ adb shell ls /system/etc/permissions | grep "usb"
com.android.future.usb.accessory.xml`

And I know as a fact that has OTG support because I used that with arduino.

This is my build.graddle (Project)
`buildscript {
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath "com.android.tools.build:gradle:4.1.1"

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        google()
        jcenter()
        maven { url 'https://jitpack.io' }
    }
}`

This is the build.gradlle (Module):
`plugins {
    id 'com.android.application'
}

android {
    compileSdkVersion 30

    defaultConfig {
        applicationId "com.smartvisioncr.ioio_otgtest"
        minSdkVersion 16
        targetSdkVersion 30
        versionCode 1
        versionName "1.0"

        testInstrumentationRunner "androidx.test.runner.AndroidJUnitRunner"
    }

    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android-optimize.txt'), 'proguard-rules.pro'
        }
    }
    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }
    buildToolsVersion '25.0.2'
}

dependencies {

    implementation 'androidx.appcompat:appcompat:1.2.0'
    implementation 'com.google.android.material:material:1.3.0'
    implementation 'androidx.constraintlayout:constraintlayout:2.0.4'
    testImplementation 'junit:junit:4.+'
    androidTestImplementation 'androidx.test.ext:junit:1.1.2'
    androidTestImplementation 'androidx.test.espresso:espresso-core:3.2.0'
    implementation 'com.github.ytai.ioio:IOIOLibAndroid:5.07'
    implementation 'com.github.ytai.ioio:IOIOLibAndroidBluetooth:5.07'
    implementation 'com.github.ytai.ioio:IOIOLibAndroidAccessory:5.07'
    implementation 'com.github.ytai.ioio:IOIOLibAndroidDevice:5.07'
}`

And the manifest:

`<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.smartvisioncr.ioio_otgtest">
    <uses-feature android:name="android.hardware.usb.accessory" />

    <application
        android:allowBackup="true"
        android:icon="@mipmap/ic_launcher"
        android:label="@string/app_name"
        android:roundIcon="@mipmap/ic_launcher_round"
        android:supportsRtl="true"
        android:theme="@style/Theme.IOIO_OTGTest">
        <uses-library android:name="com.android.future.usb.accessory"
            android:required="false" />
        <activity android:name=".MainActivity"
            android:launchMode="singleTask">
            <intent-filter>
                <action android:name="android.intent.action.MAIN" />
                <category android:name="android.intent.category.LAUNCHER" />
            </intent-filter>

            <intent-filter>
                <action android:name="android.hardware.usb.action.USB_ACCESSORY_ATTACHED" />
            </intent-filter>
            <meta-data android:name="android.hardware.usb.action.USB_ACCESSORY_ATTACHED"
                android:resource="@xml/accessory_filter" />
        </activity>
    </application>

</manifest>`

Here the complete app file if further inspection is needed:
[app.zip](https://github.com/ytai/ioio/files/6142622/app.zip)

hannesa2, 2021-03-15:
For the preparation of the issue with a running example I can only write: well done !

Unfortunately my board is broken, so I at least can't jump into this issue right now

hannesa2, 2021-03-16:
Maybe @topherbuckley can help you and make a short smoketest.

Anyway, if you find the reason, please share your solution

topherbuckley, 2021-03-17:
@techboycr I downloaded your app.zip and ran it on one of my Pixel 3as without issue. I can connect to the IOIO board and turn the LED on/off with the GUI button. 

The most common cause for not connecting is forgetting to turn off USD-Debugging after compiling+pushing to the phone. Can you verify you have turned it off?

Other than that, there are a bunch of reasons I could think of, but can you send a photo or diagram of how you wire everything up? I.e. are you powering the IOIO board externally, or expecting it to draw power from the phone (Host/Slave). Also please be specific about what kind of cables you are using between the IOIO board and phone. Are you making use of the orange usb adapter that comes with them (at least through Sparkfun) or using some other kind of micro-usb to micro-usb cable? Please clarify which micro-usb plug is on each end, etc. (ref. https://external-content.duckduckgo.com/iu/?u=https%3A%2F%2Fi.stack.imgur.com%2FpTggq.png&f=1&nofb=1) . 

For reference, in the test I ran, I had the orange usb-micro A plug attached to the IOIO board, and a standard USB-Type A to USB-C cable attached between the orange cable and my phone's USB C jack. The IOIO board was externally powered through the JST jack and the IOIO board was in A (Automatic) rather than H (Host) mode. 

See the LED ON and my wiring below:
[image]

Took another photo to show the IOIO firmware/hardware versions in the TOAST popup in case that also may be a cause. 
[image]

techboycr, 2021-03-17:
Guys I am so embarrassed. The first device I was programming over "ADB TCP" but at some point change the device and totally forgot about the usb debugging.

I so sorry that I wasted your time.

Good thing  is, I can report that works with:
Samsung A10 -->Android 10 
Xiaomi Red Mi --> Android 9

Thanks you so much for your help!

closed 2021-03-17 by techboycr