ytai/ioio #244: High latency when using Uart over USB

battery5121245, 2021-04-02:
I am developing an Android application that needs to work with a peripheral device through IOIO's UART. I am opening the UART at 115200 in the activity's setup() function, then attempting to exchange commands over the serial link.

I have noticed that responses from my device take about 50-60ms to reach the Android device. I have connected a logic analyzer to the UART pins between the device and IOIO board and any command sent by ioio takes about 2ms to get a reply from my device, however it takes another 50-60ms to reach my Android application.

This also happens if I connect the TX and RX pins on the IOIO board, basically echoing back whatever is sent out through the UART.

I am using ioio-bridge but this also happens if I connect the IOIO board directly to the Android device and use AOA.

Steps to reproduce: 
1) open an UART at 115200 bps on pins, say, 27 and 28 and physically connect them together
2) write something to the UART from the Android application, println() a timestamp in ms
3) wait for the echo, println() another timestamp
4) notice a delay of about 40-60ms between sending the message and receiving the echo.

Furthermore:
5) send random packets of 10-100 bytes repeatedly with a 1-5ms delay between sends, read the echo
6) after some time (usually under a second) the ioio board disconnects and won't reconnect
7) to reconnect to the ioio board one must restart the Android application.

Also sometimes some frames lose a few bytes between IOIO and the application, unsure whether it's the firmware or Android library or simply timeouts.

topherbuckley, 2021-04-14:
Sorry, I'm not familiar with UART but would refer you to the microcontroller's [datasheet](https://cdn.sparkfun.com/datasheets/Dev/Android/39975a.pdf) to check if your setup is within specs the ucontroller can handle. 

Also I'd note that my round trip time from java to the ioioboard is often in the range of 3 - 125 ms. (Yeah thats a huge spread) but I think this mainly has to do with other functionality the android device is dealing with (system, wifi, etc.). That being said, I think your lower bound should be an order of magnitude less.

techboycr, 2021-04-15:
Actually what @topherbuckley says is correct. Is not a problem with IOIO itself but rather a delay cause by the overall system.

The thing here is that Android is doing so much more parallel to the IOIOLooper and takes some extra ms to finish the process and that is the delay you experience.

This happened to me once similar situation not with UART but with digital input. I used and Attiny85 to interface another sensor, when the reading gets to the desired reading a digital pin goes HIGH and will be detected by another digital pin from the IOIO OTG. 

Bottom line, don't use IOIO OTG for time critical communication.