ytai/ioio #248: Implement IOIO into an existing app

techboycr, 2021-04-13:
OK ok IOIO Time!! 

If IOIO exist inside of the IOIOActivity how do I implement IOIO in an existing app that works in an activity extension of other library?

`public class CustomActivity extends AppCompatActivity implements XListener, XResultListener, XFlipListener {
//the rest comes here
}`

How should be implemented?

hannesa2, 2021-04-14:
You mean, you don't want to use `public class CustomActivity extends IOIOActivity implements ....` ?

Keep in mind it inherits from your base class `public abstract class IOIOActivity extends AppCompatActivity`

If you really want to do this (I see no point to do so) you have to copy all from `IOIOActivity` into your `CustomActivity`
The interface `IOIOLooperProvider` and the `helper_`

closed 2021-04-14 by hannesa2

topherbuckley, 2021-04-14:
I'm not quite sure what you are trying to achieve. Could you give a minimal code/psuedo code example?

What are you referring by IOIO when you say "IOIO exist inside of IOIOActivity"? 

I think @hannesa2 gave you what you need, but if not feel free to reopen the issue with additional details.

techboycr, 2021-04-15:
Thank you so much for your guidance!

Unfortunately i can share most of the code in this case. But the point is that I already have a MainActivity that extends from other library, something like:

`public class CustomActivity extends libraryXYZ implements XListener, YResultListener, ZFlipListener { //the rest comes here }`

So, for this reason I can't extend IOIOActivity. But I need activity with libraryXYZ to work with IOIO library.

At the end I am using the IOIOService example and the activity is going to talk with the service. I will let you know if this works and post an update.

topherbuckley, 2021-04-15:
Yes, I guess a service would be one way to go about it. 

Are you the author of libraryXYZ as well? If so, you can just have it extend IOIOActivity rather than Activity or AppCompatActivity. 

If thats not the case, then you could have your entry point activity be something that extends IOIOActivity, and then when finished connecting, and initializing anything else, create a new instance of your `CustomActivity` via some intent. Now that I look at it, it looks like this is what the HelloIOIOService is doing, so yes :) that sounds like a good way to go about this. 

I haven't used it like this, but [explicit creation of an IOIO instance](https://github.com/ytai/ioio/wiki/Using-AbstractIOIOActivity-(deprecated)) may also be of interest.