ytai/ioio #267: Unknown maven repos in project build.gradle

topherbuckley, 2021-08-06:
@hannesa2 do you know what [these maven repos](https://github.com/ytai/ioio/blob/fe75f00b0751ac0be23648b60db93184b0c0f302/build.gradle#L8-L9) are for? I was just going over the build files again and noticed I didn't know where these were being used. I didn't see any reference in the Github Actions, but are you publishing to these manually or elsewhere that I'm not seeing. Otherwise was planning to trim them out.

hannesa2, 2021-08-06:
Good point.
They one is obsolete #269 and the other is still used #271

```
Execution failed for task ':IOIOLibPC:compileJava'.
> Could not resolve all files for configuration ':IOIOLibPC:compileClasspath'.
   > Could not find com.sparetimelabs:purejavacomm:1.0.1.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/com/sparetimelabs/purejavacomm/1.0.1/purejavacomm-1.0.1.pom
       - https://repo.maven.apache.org/maven2/com/sparetimelabs/purejavacomm/1.0.1/purejavacomm-1.0.1.pom
       - https://jitpack.io/com/sparetimelabs/purejavacomm/1.0.1/purejavacomm-1.0.1.pom
     Required by:
         project :IOIOLibPC
```

Btw, this maintainer less and outdated `OneJar` blocks a Gradle update

```
plugins {
    id "com.github.onslip.gradle-one-jar" version "1.0.6"
}
```

Do you need it, or is something against to remove it without replacement ?

closed 2021-08-06 by hannesa2

topherbuckley, 2021-08-06:
Nice speedy work! 

From a quick log search it looks like this plugin syntax was introduced [here](https://github.com/ytai/ioio/996b01c26f0c82f89f8c34f049cc38ff06fd0afe) with the initial plugin applied [here](https://github.com/ytai/ioio/efbb065a242c816957d6eb234a4cfcfb7d771c82) way back in 2015 for "initial gradle support". 

Looks like it may have been used to create some of the files hosted in the wiki [Downloads](https://github.com/ytai/ioio/wiki/Downloads) section. I'm not planning to build things that way, but it may have been useful for building the IOIO build tools (IOIODude, IOIOBridge, etc.). Not sure if it would still be useful to you or not, but I have no plans to maintain those items, so I leave it to you.