ytai/ioio #356: Waiting for handshake

chrisws, 2023-12-28:
Hi,

I just bought an ioio-otg board for a project.

I tried the HelloWorldConsole program on various Ubuntu 22.04.3 machines. It never gets past the "Waiting for handshake" step. It works fine on Windows 10. 

I have installed the udev file. There seems to be some associated errors in the syslog. 

Are there any tricks for getting this to work in any recent linux version?

chrisws, 2024-02-24:
Never mind...

I have a solution in case anyone is interested. I switched over from using purejavacomm to fazecast.
Here's my project: https://github.com/smallbasic/smallbasic.plugins/tree/master/ioio

closed 2024-02-24 by chrisws

hannesa2, 2024-02-24:
Thank you for the solution