ytai/ioio #359: BLUETOOTH_CONNECT should be limited to SDK 30

lordfolken, 2024-05-09:
Hello, 

We use the library in XCSoar, and we get the following error in the log on modern android devices (14).
``05-09 03:48:52.944  6708  6745 E BluetoothIOIOConnection: Did you forget to declare uses-permission of android.permission.BLUETOOTH?
05-09 03:48:52.945  6708  6745 E XCSoar  : Failed to initialise IOIO bootstrap 'ioio.lib.android.bluetooth.BluetoothIOIOConnectionBootstrap@7e303b9': Need android.permission.BLUETOOTH_CONNECT permission for android.content.AttributionSource@f163109d: AdapterService getBondedDevices
``

I believe this needs to be corrected here:
https://github.com/ytai/ioio/blob/a42533b05d0776e1fb36d92680469aa5194b4016/IOIOLibAndroidBluetooth/src/main/AndroidManifest.xml#L4

and added like this here:

    
    <uses-permission android:name="android.permission.BLUETOOTH"
                     android:maxSdkVersion="30" />

hannesa2, 2024-05-09:
Btw, I'm not the only one, who can make pull requests
https://github.com/ytai/ioio/pull/365

closed 2024-05-09 by hannesa2

lordfolken, 2024-05-09:
> Btw, I'm not the only one, who can make pull requests #365

I'm sorry I know. I just wasnt sure about the correctness of it.

hannesa2, 2024-05-09:
> I just wasnt sure about the correctness of it.

Me neither. I expected, that you know more and successfully tested it. But we can give it a try.
But before, I'll fix this time thief onejar library

hannesa2, 2024-05-10:
Here you go https://github.com/ytai/ioio/releases/tag/6.3

@lordfolken please keep me informed, if it solves your issue

PR #365 (2024-05-09): Bluetooth on older devices
close #359