ytai/ioio #382: spiMaster.writeRead write buffer transaction limit

chrisws, 2024-06-05:
Is it possible to increase the SpiMaster transaction limit from 64 to 255 (as per twi/i2c) ?

In `firmware/app_layer_v1/spi.c` the TX/RX buffers are 255 - so is the limitation just in `IOIOProtocol.spiMasterRequest` ?

hannesa2, 2024-06-05:
Please make a pull request

chrisws, 2024-06-05:
Actually - I see now why there is a limit - with this:

writeByte((dataNeqTotal ? 0x80 : 0x00) | (resNeqTotal ? 0x40 : 0x00) | totalBytes - 1);

bit 7 + 8 are status bits, so there is only room for 6 bits to hold the length (63 max) 

This corresponds to the SPI_DATA_ARGS definition in firmware/app_layer_v1/protocol_defs.h

So, it would require a firmware change to resolve. For an 128x128 16 bit color LCD the buffer size needs to be a lot bigger. I was hoping there was a cheap, java side only fix to improve this. Oh well.

closed 2024-06-06 by chrisws

hannesa2, 2024-06-07:
> So, it would require a firmware change to resolve.

I've no idea how to do this. But when you know more, we could do this, if we have the source of it

chrisws, 2024-06-07:
The source code for the firmware is in the repository, but I have no idea either.

Hopefully the original author(s) could help with some expert advice.

hannesa2, 2024-06-07:
@ytai Maybe you know this ?

hannesa2, 2024-06-07:
This is the current state https://github.com/ytai/ioio/pull/191#issuecomment-2154259758