ytai/ioio #383: Implementation problem in android studio

samrakchan, 2024-06-11:
Strange issue, implementing the project directly via GitHub is not working:

```
implementation ("com.github.ytai.ioio:IOIOLibAndroidBluetooth:6.3")
implementation ("com.github.ytai.ioio:IOIOLibAndroidAccessory:6.3")
implementation ("com.github.ytai.ioio:IOIOLibAndroidDevice:6.3")
```
    
However, it works if the AAR files are placed in the lib folder:
```
    implementation files("libs/IOIOLibAndroidAccessory-release.aar")
    implementation files("libs/IOIOLibAndroidBluetooth-release.aar")
    implementation files("libs/IOIOLibAndroidDevice-release.aar")
```

samrakchan, 2024-06-11:
[image]

hannesa2, 2024-06-11:
What's the definition of "Implementation problem" ?

samrakchan, 2024-06-11:
Hi @hannesa2,

I thought I had provided enough context in my previous comments regarding the issue. Apologies if it was insufficient. Let me clarify:

I am having trouble using the dependency with this implementation:

`implementation "com.github.ytai.ioio:IOIOLibAndroidBluetooth:6.3"`

It doesn't work. However, it does work when I download the AAR file and use it like this:

`implementation files("libs/IOIOLibAndroidBluetooth-release.aar")`
Could you please check the screenshot I uploaded in my previous comment for more details?

Thank you!

hannesa2, 2024-06-12:
I see `IOIOLibCore` is not published and necessary

hannesa2, 2024-06-12:
Sorry, at least I didn't found a solution.

hannesa2, 2026-10-19:
It should work now with https://github.com/ytai/ioio/releases/tag/6.4 which includes https://github.com/ytai/ioio/pull/481
This is the test of it https://github.com/ytai/ioio/pull/388

Please reopen, if it still exists

closed 2026-10-19 by hannesa2